tart-telescope/tart2ms #6: RuntimeWarning: overflow encountered in double_scalars

maxscheel, 2022-08-17:
`!tart2ms --json tart_data.json --ms tart_data.ms`
```
Getting Data from file: tart_data.json
Writing measurement set 'tart_data.ms'...
2022-08-17 09:43:00,765 - root - INFO - Time 5167445918.181001
/usr/local/lib/python3.7/dist-packages/astropy/time/utils.py:179: RuntimeWarning: overflow encountered in double_scalars
  c = 134217729. * a  # 2**27+1.
/usr/local/lib/python3.7/dist-packages/astropy/time/utils.py:181: RuntimeWarning: invalid value encountered in double_scalars
  ah = c - abig
```

tmolteno, 2022-08-18:
This is a mysterious issue. The time logging is a valid number (corresponds to the number of seconds in the epoch). More testing will be needed.  Possibly the bug is in astropy...?

tmolteno, 2026-09-28:
Root-caused — this is an astropy bug, not a tart2ms bug, and the logged time is valid. Documented in code (commit `bb327dc`, pushed).

**The `INFO - Time 5167445918.181001` line is not a bogus time.** It is the observation epoch in seconds since the MJD epoch (1858-11-17) — exactly the MS TIME convention (casacore measure epoch value × 86400, see `timestamp_to_ms_epoch` in tart2ms/tart2ms.py). It equals MJD 59808.4018 = 2022-08-17T09:38:38 UTC, i.e. the timestamp in the input JSON; the "~163 years" is measured from the MJD epoch. (@tmolteno's comment that the time logging is a valid number is correct.)

**The warning itself** comes from `iers.conf.auto_max_age = None` (which tart2ms sets deliberately) combined with an astropy ≤ 5.0.0 bug: the first UTC↔TT conversion computes `TimeDelta(180 - np.finfo(float).max)` inside `LeapSeconds.auto_open`, and that ~1.8e308 value trips numpy's overflow warning in `astropy.time.utils.split`. Only round-off terms are affected; results are correct (same signature as desihub/specsim#119). Reproduced exactly with astropy 4.3.1 + numpy 1.21.6 running the released 0.2.1b3 code path. Fixed upstream in astropy 5.0.1 (astropy/astropy#12713); current tart2ms requires astropy ≥ 7.1, which cannot reproduce it.

Closing as fixed-upstream/documented.

closed 2026-09-28 by tmolteno