tart-telescope/tart2ms #10: Bring in the HDF files of visilbilities into a measurement set

tmolteno, 2022-10-18:
I have added the Work-In-Progress feature of using a downloaded .h5 visibility file. These contain on minute of continuous visibilities, so have much more data than the JSON returned from the RESTful API. It would be great to get them into a single measurement set, but not sure how to keep track of the moving phase center (perhaps break it up into different pointings?)

     tart2ms --hdf  ./test_data/vis_2021-03-25_20_50_23.568474.hdf --ms test.ms

bennahugo, 2022-10-18:
I could change this to the mid point. At the moment each timestamp defines a scan it is always possible to rephase and image them together though?

bennahugo, 2022-10-18:
Perhaps this should be a flag - this is easy enough

bennahugo, 2022-10-18:
My more pressing question is this is raw data. As there is
no calibrators we need baseline gain (IFR) calibration applied. I see @rubyvanrooyen is doing this with json files, so we need to put some functionality in this toolchain to form "CORRECTED_DATA" with this otherwise it will be impossible to image?

bennahugo, 2022-10-18:
Ok I put in the requested argument to do per scan phase centres (my preference -- it is always better to keep accuracy on coordinates than try to reconstitute them - therefore default)
```
2022-10-18 16:56:01     INFO    listobs::::
2022-10-18 16:56:01     INFO    listobs::::+    ##########################################
2022-10-18 16:56:01     INFO    listobs::::+    ##### Begin Task: listobs            #####
2022-10-18 16:56:01     INFO    listobs::::     listobs(vis="tart.ms/",selectdata=True,spw="",field="",antenna="",
2022-10-18 16:56:01     INFO    listobs::::+            uvrange="",timerange="",correlation="",scan="",intent="",
2022-10-18 16:56:01     INFO    listobs::::+            feed="",array="",observation="",verbose=False,listfile="",
2022-10-18 16:56:01     INFO    listobs::::+            listunfl=False,cachesize=50,overwrite=False)
2022-10-18 16:56:01     INFO    listobs::ms::summary    ================================================================================
2022-10-18 16:56:01     INFO    listobs::ms::summary+              MeasurementSet Name:  /home/hugo/workspace/tart.ms      MS Version 2
2022-10-18 16:56:01     INFO    listobs::ms::summary+   ================================================================================
2022-10-18 16:56:01     INFO    listobs::ms::summary+      Observer: Tim     Project:   
2022-10-18 16:56:01     INFO    listobs::ms::summary+   Observation: TART(24 antennas)
2022-10-18 16:56:01     INFO    listobs::MSMetaData::_computeScanAndSubScanProperties   Computing scan and subscan properties...
2022-10-18 16:56:01     INFO    listobs::ms::summary    Data records: 16560       Total elapsed time = 61.4989 seconds
2022-10-18 16:56:01     INFO    listobs::ms::summary+      Observed from   18-Oct-2022/11:27:18.3   to   18-Oct-2022/11:28:19.8 (UTC)
2022-10-18 16:56:01     INFO    listobs::ms::summary
2022-10-18 16:56:01     INFO    listobs::ms::summary    Fields: 60
2022-10-18 16:56:01     INFO    listobs::ms::summary+     ID   Code Name                RA               Decl           Epoch        nRows
2022-10-18 16:56:01     INFO    listobs::ms::summary+     0         zenith_scan_1       14:59:43.024405 -33.13.49.21187 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     1         zenith_scan_2       14:59:44.052219 -33.13.49.23469 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     2         zenith_scan_3       14:59:45.085767 -33.13.49.25764 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     3         zenith_scan_4       14:59:46.112517 -33.13.49.28043 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     4         zenith_scan_5       14:59:47.139346 -33.13.49.30323 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     5         zenith_scan_6       14:59:48.162135 -33.13.49.32594 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     6         zenith_scan_7       14:59:49.193045 -33.13.49.34884 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     7         zenith_scan_8       14:59:50.210767 -33.13.49.37144 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     8         zenith_scan_9       14:59:51.237586 -33.13.49.39425 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     9         zenith_scan_10      14:59:52.264420 -33.13.49.41706 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     10        zenith_scan_11      14:59:53.302585 -33.13.49.44012 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     11        zenith_scan_12      14:59:54.333562 -33.13.49.46303 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     12        zenith_scan_13      14:59:55.359164 -33.13.49.48581 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     13        zenith_scan_14      14:59:56.370443 -33.13.49.50828 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     14        zenith_scan_15      14:59:57.398975 -33.13.49.53114 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     15        zenith_scan_16      14:59:58.424633 -33.13.49.55393 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     16        zenith_scan_17      14:59:59.459732 -33.13.49.57694 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     17        zenith_scan_18      15:00:00.486690 -33.13.49.59977 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     18        zenith_scan_19      15:00:01.509949 -33.13.49.62251 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     19        zenith_scan_20      15:00:02.535081 -33.13.49.64530 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     20        zenith_scan_21      15:00:03.563248 -33.13.49.66816 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     21        zenith_scan_22      15:00:04.588546 -33.13.49.69096 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     22        zenith_scan_23      15:00:05.623978 -33.13.49.71398 J2000          276
2022-10-18 16:56:01     INFO    listobs::ms::summary+     23        zenith_scan_24      15:00:06.645474 -33.13.49.73670 J2000          276
[... cut ...]
2022-10-18 16:59:42     INFO    listobs::ms::summary+      ID= 20-21: 'ANTENNA-20'='Rhodes University, Grahamstown', 'ANTENNA-21'='Rhodes University, Grahamstown', 
2022-10-18 16:59:42     INFO    listobs::ms::summary+      ID= 22-23: 'ANTENNA-22'='Rhodes University, Grahamstown', 'ANTENNA-23'='Rhodes University, Grahamstown'
2022-10-18 16:59:42     INFO    listobs::::     ##### End Task: listobs              #####
2022-10-18 16:59:42     INFO    listobs::::+    ##########################################
```

tmolteno, 2022-10-19:
This is fantastic. I think we can close this issue and mark it done! Thanks Ben.

closed 2022-10-19 by tmolteno