tart-telescope/tart2ms #11: Register the various TARTs with JPL

bennahugo, 2022-10-18:
At the moment TART is an unknown telescope. As far as I can tell the telescope must be registered with JPL. I'm not sure whether this is jumping the gun, but I'm also not sure what will fall over if we don't do this. Will need to play

tmolteno, 2022-10-18:
Reading the code. It may be that the observatory position needs to be in the metadata stored in the MS?

bennahugo, 2022-10-18:
hmm as far as I understood this was referring to the Observatories ephem table (customizable via .casarc) ... I might be wrong though. Let me have a look.

bennahugo, 2022-10-19:
I can ask Brian Butler how to go about this. He has done this for MeerKAT in the AIPS version of the CASA tables through JPL. We should decide on a telescope naming / coding convention before there are too many tarts deployed for this purpose. There also needs some facility in tart2ms to give the correct coding (perhaps it should really be based on earth location -- that will make it completely unambiguous. Lets discuss this at the meeting

tmolteno, 2026-09-28:
Status summary (2026): this is still open and the remaining work is **administrative**, not code.

Where things stand: casacore's `MSMetaData::getObservatoryPosition` / the Observatories ephemeris table don't know "TART" (setting TELESCOPE_NAME to e.g. `kat-7` makes the error disappear, which is why the workaround exists). Since the thread was last active, ~10 TARTs are deployed and some CASA tools (`plotant`) still require a known observatory name.

What remains:
1. agree a naming/coding convention for the TARTs (earth-location-based codes would be unambiguous);
2. contact Brian Butler / JPL (who did MeerKAT's AIPS/CASA registration) to get the observatory codes into the AIPS/CASA observatory table and/or the casacore Observatories meas table;
3. then bake the registered names into tart2ms defaults per site.

Meanwhile `--telescope_name` (tart2ms/scripts/tart2ms.py) already lets users override TELESCOPE_NAME. A small code-side nicety would be defaulting it from the site in the telescope database — happy to do that once the convention is settled.