tart-telescope/tart2ms #12: Antenna coordinates are invalid

bennahugo, 2022-10-18:
This is what the array looks like...
[image]

bennahugo, 2022-10-18:
> [image: image]
> <https://user-images.githubusercontent.com/85630/196478673-c887abd4-38a4-4bb5-b9ef-5c85bd5b4ddc.png>
>
> How are you getting thousands of km? The coordinates are all less than 4m?

This explains it though - these coordinates are supposed to be ITRF
coordinates. I'm guessing things fall apart when they are converted to WGS84

tmolteno, 2022-10-18:
Will use astropy conversions to do this properly. May take a few days as its getting late here in NZ. WGS84 is supported apparently according to the memo.

bennahugo, 2022-10-18:
ok no problem. I will continue trying to make things more conformant. I think first order we just need to add offsets off the array centre ITRF coordinate?

tmolteno, 2022-10-18:
This should do. It is simply location.to_geodetic() in the codebase... (location) is the tart position in astropy.

bennahugo, 2022-10-18:
Tagging @rubyvanrooyen thoughts on the proposed fix?

bennahugo, 2022-10-18:
I guess the other solid way of testing this is to regenerate UVW coordinates in casa - if we get handedness wrong it should surely show up in the direction vectors... but this is just a thought

PR #14 (2022-10-18): Issue 12
Closes #12 

This is an implementation of the proposed antenna position fix. Things definitely looks a lot better on a plot
[image]

@tmolteno @rubyvanrooyen @o-smirnov please advise? I'm unsure of the exact layout -- I presume the antenna coordinates
are the fitted coordinates relative to info dict lat lon alt? However it is anybody's guess which way the handedness were fitted.
We need somehow to establish that East and North is the right way here. I'm thinking regenerating UVW coordinates using CASA?

Note I had to put in a kludge to override the observatory name (you can specify e. [... cut]
closed 2022-10-18 by tmolteno