tart-telescope/tart2ms #27: CI testing

bennahugo, 2022-10-20:
Various workflows are available including direct download of antenna, gain and importing and exporting options

We really do need more test coverage to ensure all of this continues to work when more fields are being added to json / h5 which seems also unversioned to me.

@tmolteno any plans to unify the repositories under tart-telescope. I see half the stuff lives in your user and the other half lives here. It would be a good idea to have them all in one organization so we can more easily set up CI workflows

tmolteno, 2022-10-20:
This is a work in progress at the moment. Shifting stuff towards tart-telescope is the plan. We will keep the h5 and json unversioned as long as possible, but eventually it will come.

tmolteno, 2022-10-27:
At the moment testing is done using. Tests are currently failing.

```make test```

This runs pytest3

closed 2022-10-27 by tmolteno