tart-telescope/tart2ms #29: Rotation in plane of the sky

bennahugo, 2022-10-20:
It is really hard to tell with the tart but it seems there is a rotation in the plane of the sky -- it is most apparent when rephased to SCP -- everything seems to drift circularly around the pole (you can see some satellites moving in other directions though) which indicates to me that we have a severe problem with our hour clock @tmolteno.

bennahugo, 2022-10-20:
for information: a clock error (Hour hand) will cause a pure rotation when viewed from the pole (there are no uv projection effects then. See e.g. the DEEP2 paper by Mauch et al. on early MeerKAT days.

bennahugo, 2022-10-20:
I will try make a little movie to show what I'm talking about

bennahugo, 2022-10-20:
https://drive.google.com/file/d/1M59SK3Wx6O1moNhuwZH_pNL5LvdAxsPb/view?usp=sharing

tmolteno, 2022-10-24:
Attached is the a directory of files that can be used to make a movie... These are taken every two minutes for an hour...

https://drive.google.com/drive/folders/1YiJw5nZRvPp_Euc3IrJz1tbJKFVAizGe?usp=sharing

There is a Makefile that will build a movie from the JSON files that seems to work.

https://user-images.githubusercontent.com/85630/197627961-ed6263aa-ecd4-4ed4-9862-62fc6a3d891d.mp4

tmolteno, 2022-10-24:
Here's the wsclean video generated from the MS created from the JSON files (no rephasing)

https://user-images.githubusercontent.com/85630/197651482-cd9a754a-a861-438c-b724-d64b75170b4e.mp4

bennahugo, 2022-10-25:
Thanks @tmolteno I've downloaded the data just haven't yet been able to get to it. The issue is most apparent when you look from the polar perspective. 

https://drive.google.com/file/d/1M59SK3Wx6O1moNhuwZH_pNL5LvdAxsPb/view?usp=sharing

bennahugo, 2022-10-25:
Rephasing to polar view and then imaging with your software with the GNSS coordinates overlayed should clarify whether this is a global rotational effect or just a change in the noise structure due to PSF sidelobes -- there is really is no way to tell without known high SNR sources

bennahugo, 2022-10-25:
@tmolteno Ok I'm pretty convinced the astrometry is correct now on the non phased case (per dump zenethals). I will submit a fix to disko soon but you do appear to be using the wrong convention inside your imager. With the UVW coordinates generated using casacore the images are flipped (don't worry MeerKAT had this the wrong way around for a while even into science operation which required all users to run casa fixvis on datasets). 

When I inject conjugate before your solvers I get precisely what I want from the ms visibilities:
[image]

[image]

We now have a problem though. I'm going to say we have to accept the casacore definitions of UVW handedness (ie. the NRAO definition) otherwise all of radio astronomy is going to fall apart, however your telescope needs to write -UVW from now on end to make this conform? I tried loading in json data and ms converted data and they all come out the same way though so not sure how these json databases were written that you gave me.

bennahugo, 2022-10-25:
So I have put in a fix -- when going from json we use "telescope convention (ie the wrong way around)". When going from ms we use standard NRAO convention. I'm going to push my fixes for disko
JSON in "telescope convention"
[image]
MS in "NRAO convention"
[image]

The noise is different (due to the unholy kludge in tart2ms which we really must fix), but in zenethal mode everything lines up astrometry wise

bennahugo, 2022-10-25:
I'm going to accept the astrometry on the non-zenethal cases, because the operator we apply is absolutely box standard (same as we have in ddfacet / perley polyhedron grinding in africanus etc. and I already checked against chgcentre from Andre Offringa.

closed 2022-10-25 by bennahugo