tart-telescope/tart2ms #35: RuntimeError: Cannot convert baseline to uvw: frame information missing

maxscheel, 2022-10-29:
Example here: https://colab.research.google.com/drive/1vk2DFxLsJOPiLmGSWIio7jOZ_n8tU1Ay#scrollTo=MsitI9WJQcLH
`
tart2ms --json json/vis_2022-10-29T05:15:00.475Z.json --telescope_name 'kat-7'
`

```
Getting Data from file: ['json/vis_2022-10-29T05:15:00.475Z.json']
Writing measurement set 'tart.ms'...
2022-10-29 06:55:37,216 - root - INFO - Will process JSON file: 
2022-10-29 06:55:37,216 - root - INFO - 	 'json/vis_2022-10-29T05:15:00.475Z.json'
Processing JSON database ▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣ 100%2022-10-29 06:55:37,219 - root - INFO - <Done>
2022-10-29 06:55:37,219 - root - INFO - Observatory parameters:
2022-10-29 06:55:37,219 - root - INFO - 	L0_frequency             : 1571328000.0
2022-10-29 06:55:37,220 - root - INFO - 	bandwidth                : 2500000.0
2022-10-29 06:55:37,220 - root - INFO - 	baseband_frequency       : 4092000.0
2022-10-29 06:55:37,220 - root - INFO - 	operating_frequency      : 1575420000.0
2022-10-29 06:55:37,220 - root - INFO - 	name                     : Signal Hill - Dunedin
2022-10-29 06:55:37,220 - root - INFO - 	num_antenna              : 24
2022-10-29 06:55:37,220 - root - INFO - 	sampling_frequency       : 16368000.0
2022-10-29 06:55:37,220 - root - INFO - Telescope position (WGS84):
2022-10-29 06:55:37,220 - root - INFO - 	Lat -45.85177
2022-10-29 06:55:37,220 - root - INFO - 	Lon 170.5456
2022-10-29 06:55:37,220 - root - INFO - 	Alt 270.0
2022-10-29 06:55:37,223 - root - INFO - Min time: 2022-10-29 05:15:00.475000+00:00 -- 5173737300.475
2022-10-29 06:55:37,223 - root - INFO - Max time: 2022-10-29 05:15:00.475000+00:00 -- 5173737300.475
2022-10-29 06:55:38,161 - root - INFO - Pol Feeds ['RR']
2022-10-29 06:55:38,239 - root - INFO - Setting phase center per instantaneous-zenith
2022-10-29 06:55:38,242 - root - INFO - Writing MS for telescope name 'kat-7'
2022-10-29 06:55:38,256 - root - INFO - Baseline lengths:
2022-10-29 06:55:38,256 - root - INFO - 	Minimum: 0.0033 m
2022-10-29 06:55:38,256 - root - INFO - 	Maximum: 2.3396 m --- 12.2946 wavelengths
2022-10-29 06:55:38,256 - root - INFO - Appoximate unweighted instrument resolution: 341.1304 arcmin
2022-10-29 06:55:38,257 - root - INFO - Data column size 0.00 MiB
2022-10-29 06:55:38,258 - root - INFO - No rephasing requested - field centers left as is
2022-10-29 06:55:39,084 - root - INFO - Computing UVW coordinates for output dataset... WAIT
Calculating UVW ▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣▣ 100%2022-10-29 06:55:39	WARN	MeasIERS::findTab (file /code/measures/Measures/MeasIERS.cc, line 389)	Requested data table IERSeop97 cannot be found in the searched directories:
2022-10-29 06:55:39	WARN	MeasIERS::findTab (file /code/measures/Measures/MeasIERS.cc, line 389)+	/usr/share/casacore/data/ephemerides/
2022-10-29 06:55:39	WARN	MeasIERS::findTab (file /code/measures/Measures/MeasIERS.cc, line 389)+	/usr/share/casacore/data/geodetic/
2022-10-29 06:55:39	WARN	MeasIERS::findTab (file /code/measures/Measures/MeasIERS.cc, line 389)	Requested data table IERSpredict cannot be found in the searched directories:
2022-10-29 06:55:39	WARN	MeasIERS::findTab (file /code/measures/Measures/MeasIERS.cc, line 389)+	/usr/share/casacore/data/ephemerides/
2022-10-29 06:55:39	WARN	MeasIERS::findTab (file /code/measures/Measures/MeasIERS.cc, line 389)+	/usr/share/casacore/data/geodetic/
2022-10-29 06:55:39	INFO	MeasIERS::fillMeas(MeasIERS::Files, Double) 	Requested JD 59881.2 is outside the range of the IERS (Earth axis data) table.
2022-10-29 06:55:39	INFO	MeasIERS::fillMeas(MeasIERS::Files, Double) +	Calculations will proceed with less precision
2022-10-29 06:55:39	WARN	MeasIERS::findTab (file /code/measures/Measures/MeasIERS.cc, line 389)	Requested data table TAI_UTC cannot be found in the searched directories:
2022-10-29 06:55:39	WARN	MeasIERS::findTab (file /code/measures/Measures/MeasIERS.cc, line 389)+	/usr/share/casacore/data/ephemerides/
2022-10-29 06:55:39	WARN	MeasIERS::findTab (file /code/measures/Measures/MeasIERS.cc, line 389)+	/usr/share/casacore/data/geodetic/
2022-10-29 06:55:39	SEVERE	MeasTable::dUTC(Double) (file /code/measures/Measures/MeasTable.cc, line 4283)	Cannot read leap second table TAI_UTC
Traceback (most recent call last):
  File "/usr/local/bin/tart2ms", line 63, in <module>
    ms_from_json(ARGS.ms, ARGS.json, ARGS.pol2, ARGS.phase_center_policy, ARGS.override_telescope_name)
  File "/usr/local/lib/python3.7/dist-packages/tart2ms/tart2ms.py", line 912, in ms_from_json
    override_telescope_name=override_telescope_name, uvw_generator=uvw_generator)
  File "/usr/local/lib/python3.7/dist-packages/tart2ms/tart2ms.py", line 687, in ms_create
    fixms(ms_table_name)
  File "/usr/local/lib/python3.7/dist-packages/tart2ms/fixvis.py", line 314, in fixms
    ack=ack)
  File "/usr/local/lib/python3.7/dist-packages/tart2ms/fixvis.py", line 190, in synthesize_uvw
    posunits[2])))
  File "/usr/local/lib/python3.7/dist-packages/casacore/measures/__init__.py", line 557, in touvw
    m = _measures.uvw(self, v)
RuntimeError: Cannot convert baseline to uvw: frame information missing
```

tmolteno, 2022-10-29:
This appears to be a JSON file created with tart_vis2json. 

```
    tart_download_data --vis --n 1
    tart_vis2json --vis xxx.hdf
    tart2ms --json json/vis_vis_2022-10-29T20:29:59.372Z.json
```

bennahugo, 2022-10-31:
The issue is in the log 2022-10-29 06:55:39	SEVERE	MeasTable::dUTC(Double) (file /code/measures/Measures/MeasTable.cc, line 4283)	Cannot read leap second table TAI_UTC

You are missing ephemeris data. You will have to pull this directly from the NRAO 
mkdir -p /usr/share/casacore/data/
cd /usr/share/casacore/data/
rsync -avz rsync://casa-rsync.nrao.edu/casa-data .

tmolteno, 2022-10-31:
I think there is a command installed on debian as part of casacore
```
    casacore-update-tai_utc
```
which seems to do this. Perhaps

maxscheel, 2022-10-31:
```
!casacore-update-tai_utc -f -o/usr/share/casacore/data/geodetic/TAI_UTC
```
Did the trick!

closed 2022-10-31 by maxscheel