tart-telescope/tart2ms #44: --hdf-archive option

tmolteno, 2022-12-07:
parser.add_argument('--archive-search', required=False, default=None, nargs="*",
                        help="Query hdf5 files from the TART online archive. (-10:1:0 last ten minutes interval 1 min) -100:10:0 last 100 minutes every 10 minutes. START:INTERVAL_MIN:END arbitrary timestamps")

bennahugo, 2022-12-09:
I think it is actually better to specify UTC start and end times if possible and/or LST spans

tmolteno, 2022-12-11:
The major use will be to get something like the last hour of data, and this will make the scripting much more intuitive. 

    signal:-120:1:-1 
    
We intend to have <START> and <END> be specified initially as unix timestamps if they are positive.  Adding ISOFORMAT dates with UTC specified can be addressed with a separate PR once this is done.

closed 2026-09-27 by tmolteno

tmolteno, 2026-09-28:
Implemented in commit `a3305dd` (pushed; auto-closed by the commit message). New CLI option:

    tart2ms --archive-search "<Name>:START:INTERVAL:END" --ms out.ms --uncalibrated

It queries the TART S3 archive (via tart_tools' `archive_handler`) and converts the matched HDF5 files in one go. START/END accept minute offsets from now (`signal:-10:1:0` = last ten minutes, every minute) or ISO-8601 timestamps for arbitrary windows; INTERVAL is the sampling interval in minutes. Verified end-to-end against the live bucket: `nz-dunedin:2026-06-01T00:00:00:10:2026-06-01T01:00:00` downloaded 7 files and produced a 115,920-row MS.

Notes for anyone using it:
- the bucket telescope names have changed since this issue was filed (now `nz-dunedin`, `nz-elec`, `za-rhodes`, `za-hammanskraal`, `mu-udm`, `na-unam`, ... — the old `signal/rhodes/stellenbosch` prefixes no longer exist);
- queries must include the `<Name>:` prefix (a leading `-` on a bare offset would confuse argparse);
- INTERVAL selection is approximate (evenly spaced picks); requires `minio` (clear error if missing); sparse archive windows warn/raise gracefully.