tart-telescope/tart2ms #46: Rephase to Sun / Moon / Sgr A*

bennahugo, 2022-12-09:
Add special keywords to rephase to these positions

tmolteno, 2022-12-11:
I think we need to resolve issue #47 first so see how arbitrary places are specified. Then this becomes a very easy option to add later. 

These are potentially more useful than NCP and SCP which aren't likely to be used for deep sky work!

bennahugo, 2022-12-19:
Not sure why not though - they are as good as any other celestial position. I have left them as special positions, which are defined as follows:
```json
{
        "name": "SCP",
        "position": {
            "RA": "00h00m00s",
            "DEC": "-90d00m00s",
            "EQUINOX": "J2000",
            "FRAME": "fk5"
        }
}
```
along with positions for e.g. SgrA*

bennahugo, 2022-12-19:
I will implement solar system bodies now - these require special ephemeris to be computed - the written FITS ephemeris is therefore arbitrary (RA / DEC is ill defined in that frame.

PR #48 (2022-12-30): Celestial catalogs, arbitrary celestial phasing, solar system phasing and minor bug fixes
Closes #46 
Closes #47 
Closes #45 

Implements phasing (fringe stopping) for:
 - arbitrary user position in J or B twelve digit format as requested by @tmolteno 
 - Known solar system objects (e.g. Sun, Moon)
 - Known extragalactic fields
 - Some specifyable field aliases via json
 - Major performance optimization in writing ::SOURCE table
 - Major graph performance optimizations for solar system phasing
 - some cleanup and minor fixes in previous phasing implementation for no-phasing option
closed 2022-12-30 by tmolteno