tart-telescope/tart2ms #47: Rephase to arbitrary user provided position

bennahugo, 2022-12-09:
Needed to rephase to e.g. Fornax A for mapping

tmolteno, 2022-12-11:
Agree that this needs to be done. I think adding a

  --rephase {obs-midpoint, ICRS}
                        Rephase all visibilities to a new phase center (default: None)

and removing the SCP and NCP (since these are just special cases of ICRS coordinates). We need to be absolutely specific about which reference frame we are using e.t.c. if it is to be ICRS, we need to point to a reference document about how to generate these co-ordinates. What document (URL) do you propose?

bennahugo, 2022-12-12:
I think I will keep all the options under one CMD flag - it is useful to
recognise names from catalogs, special bodies and special positions. The
solar system objects can only be tracked by name as they don't have a fixed
ephemeris.

tmolteno, 2022-12-12:
I was referring to getting rid of NCP and SCP since these are probably not targets to track. Having some objects by name isn't a bad idea, but I this issue is around specifying arbitrary positions, and so we need to be absolutely specific about which reference frame we are using . We therefore need to point to a reference document about how to generate these co-ordinates. Which celestial frame are you propose? 

We then need to point to a reference document about how a user can generate these coordinates. This should probably be in  the [docusaurus] (https://github.com/tart-telescope/tart_website), with references. 

We can decide which objects to track by name in Issue #46, but here we should make the decision about which co-ordinate system and syntax for specifying it should be used.

tmolteno, 2022-12-12:
I have added a placeholder at (https://github.com/tart-telescope/tart_website/blob/main/docs/tutorial-extras/measurement-sets.md) which can point to documentation about the co-ordinates and syntax for them.

bennahugo, 2022-12-12:
Not sure I want to remove the poles as named positions, rather just augment
with additional recognized objects and positions. I'm inputting 3CRR
catalogs with names as filed. I will perhaps move them to a json somewhere
and document them as requested

I think we should stick with J2000 positions throughout and use regex to
recognize positions. So midpoint, ncp, scp are recognized as stands (plus
and additional named positions to be read). Then special objects (Sun,
Moon, Mars etc). Then catalog names from the 3CRR catalog (ASCII catalog as
available online), then regex extracted positions J2000 as specified by casa

tmolteno, 2022-12-12:
This issue is about specifying celestial coordinates. Tracking objects by name is Issue #46.

1. We have to choose a syntax for coordinates. This is simply not optional.
2. We must provide a link to where this is defined (first in this issue, then in the code and also in the documentation), and how they are going to be represented. 
3. Technically  J2000 is a reference epoch. (https://en.wikipedia.org/wiki/Epoch_(astronomy)#J2000), not a coordinate system.  I suggest we use ICRS.
4. Here is the astropy documentation (https://docs.astropy.org/en/stable/coordinates/index.html). Exactly which syntax and grammar are going to be used for coordinates on the command line?

bennahugo, 2022-12-13:
@tmolteno I agree we should use ICRS as the system, but to be honest, there is no difference between FK5 and ICRS for the resolutions we are talking about (order few tens of mas whereas our resolution is at best degrees...). There is no need to overcomplicate the CMD arguments with additional system frame specifications, e.g. epoch ra dec as follows
```
J2000 9h25m00s -05d12m00s
```

the syntax will therefore be
--rephase {obs-midpoint, ncp, scp, [solar system bodies], [known 3GRR positions], radec} with appropriate documentation in the CMD with examples.

tmolteno, 2022-12-14:
We need a no_space format for the arbitrary positions. The new IAU name - which corresponds to the co-ordinates is ideal for this purpose:

```
    tart2ms --rephase J0925-0512
```

This is both sufficiently accurate, from a TART resolution point of view, as well as allows people to use source designations from papers which commonly use the new IAU format.

Notes:

- This would mean that NCP is J0+90, and SCP is J0-90.
- Higher accuracy might need more precision, but suggest doing this with a different PR.
- I think this is pretty friendly, and would mean we don't need to try and build fixed source catalogues into tart2ms, but rather can read any catalogue (https://3crr.extragalactic.info/cgi/database) and easily enter the designator.
- @rubyvanrooyen might have useful input here.

bennahugo, 2022-12-15:
Excuse the brevity, not at my desk. Reason for the more elaborate format is
to specify B1950 epoch and GAL coordinates.

tmolteno, 2022-12-18:
This does not preclude a more general method, however I think that this should not be done until a proper syntax is defined. At this stage:

- this accomplishes almost every use case I can imagine, 
- anyone alternative coordinate systems will have the skills to convert them.
- It is easy to document and understand.

The key is to have code that is easy to maintain, and this means doing the minimum necessary to do the job.

bennahugo, 2022-12-19:
I leave SCP and NCP along with arbitrary specifyable lists of special coordinates and catalog defined coordinates as part of the options

PR #48 (2022-12-30): Celestial catalogs, arbitrary celestial phasing, solar system phasing and minor bug fixes
Closes #46 
Closes #47 
Closes #45 

Implements phasing (fringe stopping) for:
 - arbitrary user position in J or B twelve digit format as requested by @tmolteno 
 - Known solar system objects (e.g. Sun, Moon)
 - Known extragalactic fields
 - Some specifyable field aliases via json
 - Major performance optimization in writing ::SOURCE table
 - Major graph performance optimizations for solar system phasing
 - some cleanup and minor fixes in previous phasing implementation for no-phasing option
commit 2b6072b6 (2022-12-30): Fixes catalog position readings. Closes #45
Implement arbitrary rephasing. Closes #47
Part work towards #46 - Special bodies still to get proper UVW coordinates and phases,
but the user can now put in arbitrary special positions. Pole positions are moved to
this system.
closed 2022-12-30 by tmolteno