tart-telescope/tart2ms #50: Astrometry errors after rephasing

bennahugo, 2023-01-06:
Bit trial by error but I think I finally tracked down the astrometry issues I'm seeing on a long track. it is in my phasing code -- it is a matter or signage convention for the uvw. The top left is the last zenith dump of this particular field (model visibilities shown). The others are various options on the phasing where the field is rephased to the mid zenith point on the two hour scan. The bottom one matches 1:1. I will submit a fix for this soon

[image]

_Originally posted by @bennahugo in https://github.com/tart-telescope/tart2ms/issues/49#issuecomment-1374126620_

bennahugo, 2023-01-06:
See also
https://github.com/tart-telescope/tart2ms/pull/49#issuecomment-1374173016

tmolteno, 2026-09-28:
The promised fix landed in PR #49 ("Model predict", merged 2023-01-11): commits `f5619648` ("Fix astrometrical issues in rephasing introduced when solar rephasing was added") and `e89e38b6`. Both are ancestors of master.

The root cause turned out to be **not** the uvw signage convention but zenith/field indexing when solar-system tracking was added (`refdir[ifid]` → `refdir[fid]` in `rephase()`), plus the exact spherical n-term and an fk5→icrs field-frame fix. All of it is in master today (tart2ms/fixvis.py:250, 256, 266-268, 278-281), and @bennahugo verified it in the PR thread ("Normality is restored... Astrometry working as it should - calibrated data :)"). He typed "Closes #50" in a comment — which doesn't close issues, so this has just been left open by oversight.

Closing — please reopen if anyone still sees astrometric offsets with current master.

closed 2026-09-28 by tmolteno