tart-telescope/tart2ms #52: Fails on python 3.11 due to numba error.

tmolteno, 2023-01-30:
May have to wait on this one. It is installed as a dependency of codex-africanus. 

`  Downloading numba-0.56.4.tar.gz (2.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.4/2.4 MB 1.9 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-zfdfkg4q/numba_1d9792cd9fdb44dda13fea8a2a0ebaef/setup.py", line 51, in <module>
          _guard_py_ver()
        File "/tmp/pip-install-zfdfkg4q/numba_1d9792cd9fdb44dda13fea8a2a0ebaef/setup.py", line 48, in _guard_py_ver
          raise RuntimeError(msg.format(cur_py, min_py, max_py))
      RuntimeError: Cannot install on Python version 3.11.1; only versions >=3.7,<3.11 are supported.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.
`

tmolteno, 2023-01-30:
This is a problematic one. Numba is produced by anaconda and they're waiting on official anaconda packages (and numba messes with python bytecode). It would be really good to get rid of this dependency. @bennahugo I think the codex-africanus dependency came in with some of the rephasing code. Is it really needed? According to the numba bug report, every new version of python will cause a 3-6 month delay in numba (https://github.com/numba/numba/issues/8304)

bennahugo, 2023-01-31:
The numba dependency came in with the DFT prediction code needed to calibrate TART data (the kernels are written using Numba JIT - which is easier to maintain (and compile for the average user) than C++ kernels. Maybe @sjperkins can comment but I don't really see a way around this (aside from rewriting a portion of tart2ms to compile some C++17 bindings.

sjperkins, 2023-01-31:
Hello @tmolteno. I want to understand more of the context here. Python 3.11 is fairly bleeding edge (released 24 Oct 2022). What are the reasons for requiring your software to run on it? Would it be possible to run on Python 3.10 instead?

tmolteno, 2023-01-31:
Hi @sjperkins ! We run debian testing as standard on all development machines. This has introduced a huge dependency chain that (for example) means we can't run on standard raspberry pi OS 32-bit. 

@bennahugo I have tried disabling numba using  [NUMBA_DISABLE_JIT](https://numba.readthedocs.io/en/stable/reference/envvars.html#envvar-NUMBA_DISABLE_JIT) and see no significant performance changes using the following command:

```
        tart2ms --hdf ~/Downloads/tart_obs/tart_obs/vis_2022-10-21_10_19_46.661434.hdf \
                --rephase obs-midpoint --single-field \
                --ms test_data/test.ms --clobber
```
Which command line options need this dependency?

bennahugo, 2023-02-01:
You may not see significant degradation on a single timestamp, but when
combining many databases (10s of minutes to hours it starts hitting in the
synthesizing ms step - which is where the predict gets executed).

bennahugo, 2023-02-01:
Also just a note that you can disable model predict, which will then not
require codex - the import is already guarded for this reason. Perhaps this
can be made an optional dependency  with the square bracket notation for PI
deployment if there are no 32bit wheels available - not sure because I
don't have such a system at hand to test with?

 However the data tart2ms then produce is of limited value because the
phases are all jumbled, but perhaps this is all you want for the pi
deployment is basic conversion of uncalibrated data?

For general telescope user deployment I do think a first order model_data
column is necessary if you want any chance of imaging.

tmolteno, 2023-02-01:
Good to know. The data from a calibrated telescope is not jumbled. I think one solution is to move the codex-africanus dependent prediction code into a tool specifically for this job. Then we get the best of both worlds. The construction of measurement sets is separated from their post-processing.

commit dfbfb0fc (2023-02-01): Fix tests to only test prediction if codex-africanus is installed. Closes #52
closed 2023-02-01 by tmolteno