tart-telescope/tart2ms #53: Ditch Astropy for Skyfield

bennahugo, 2023-02-02:
According to testing @jskenyon has been doing on unrelated work the equatorial to local transformations in Skyfield is at least an order of magnitude faster. Curently this is one of the remaining bottlenecks in selecting sources (working out their az el etc. -- this is slower than the actual predict which is done via JIT code). 

I feel strongly about ditching astropy so that we can make the codebase as fast as possible to stitch together many days of observation for scientific analysis and classification of the system SEFD once we can pick up astronomical sources like fornax.

tmolteno, 2023-02-16:
This would depend a bit on what dependencies skyfield brings in. If it's numba, then we probably can't because of Issue #52

tmolteno, 2026-09-28:
Assessed against current master (2026). Agree with the goal; the scope is narrower than it looks, but a full swap is still not the right first step.

**Where astropy is hot in source selection:** `radec2azel` / `azel2radec` / `get_solar_system_bodies` / `get_catalog_sources_azel` (tart2ms/ms_helper.py), the per-timestamp zenith transform (tart2ms/tart2ms.py), and `get_body` calls. Skyfield would cover exactly those. Note the issue thread's numba concern does not apply: Skyfield's dependencies are just certifi/jplephem/numpy/sgp4.

**But** astropy is also load-bearing for phase-centre parsing (named/12-ball targets via SkyCoord), units, Time/IERS configuration, and antenna placement — and the l,m,n ↔ uvw frame convention (see #56) must stay consistent with casacore. A full removal is a multi-day, frame-risky refactor.

**Recommendation:** (1) re-profile first — the 2026 work (per-source-epoch memoization of `predict_model`, direct RA/Dec from the catalogue client) already cut much of the astropy load in the hottest loop, so the 2023 "order of magnitude" numbers may no longer apply; (2) if transforms still dominate, do a targeted Skyfield fast-path for the three transform helpers + Sun/Moon (~1 day plus regression tests) and keep astropy for parsing/frames. Keeping this open as an enhancement.

tmolteno, 2026-09-28:
Re-profiled on master @ e4864c8 (py3.13, astropy 8.0.1, skyfield 1.55), shipped test data (60 timestamps × 276 baselines), medians of 3 on a shared sandbox CPU. Scripts and raw pstats in [`.work/profile/`](https://github.com/tmolteno/tart_modules) (local; happy to add them to the repo).

**TL;DR:** transforms still dominate the `--add-model` path (80–90%), but the cause is 5760 *scalar* per-source `SkyCoord.transform_to` calls in `predict_model` (10.3 ms each ≈ 59 s of an 81.5 s run), not "astropy is slow". Batching them beats or matches a skyfield swap: astropy-only batching → ~3.6× end-to-end (bit-identical, no new dep), skyfield swap + batching → ~4.3× (predicted), naive skyfield swap keeping call shapes → 2.5–3.0× (**measured** 31.9 s vs 81.5 s via monkeypatched helpers). Plain conversions don't need any of this (transforms ≈10%, max 1.09×).

**Wall (60 ts × 276 bl):** plain 3.2 s warm / 4.9 s cold; `--add-model` 81.5 s warm / 95 s cold. One-off astropy IERS parse ≈1.4 s/process.

**Model-path cProfile:** `azel2radec` 108 s / 154 s (70%; 5760 scalar calls — 94 sources × 60 epochs + Sun/Moon); `celestial_positions` 18 s (12% — it does a *scalar astropy ITRS→ICRS per satellite*); `get_solar_system_bodies` 7.8 s (5%); `get_catalog_sources_azel` 6.9 s (4.5% — 2.5 s of that is re-reading the catalog files, the transform itself is 0.05 s); `wsclean_predict` 2.1 s. Plain path: casacore/dask 48%, astropy 10%.

**astropy vs skyfield, like-for-like** (no refraction either side):
- azel→radec scalar (today's call shape): 10.0 vs 1.44 ms/call — skyfield 7×
- azel→radec per-epoch vectorized: 10.5 vs 1.5 ms/epoch — skyfield ~7×; and *either* vs today's scalar loop: 90–3000×
- radec→azel vectorized (`get_catalog_sources_azel` shape): astropy 34 ms vs skyfield 343 ms (100×60) — **astropy 10–14× faster** (skyfield's `Star.observe` can't broadcast T×N; loops per timestamp)
- Sun/Moon: 3.6 s vs 0.025 s (T=60) — skyfield 146×
- zenith: skyfield 1.4–3.6×
- correctness: radec↔azel agrees 0.2–2″; absolute azel→radec differs ≤20.5″ (annual-aberration convention: skyfield `from_altaz` is rotation-only, astropy inverts aberration — pick ONE convention per direction pair, or ~0.1–0.7 rad phase shifts on the longest baselines); Sun/Moon az/alt ≤0.7″, distance ≤3e-8 rel. Gotcha: `azel2radec(distance=...)` returns **barycentric** directions for Sun/Moon (157° off topocentric!) — any rewrite must replicate that.

So the 2023 "order of magnitude" claim is **true only for scalar call shapes (7×) and Sun/Moon (146×); false for the vectorized radec→azel path (astropy is 10–14× faster there)**.

**Amdahl on the 81.5 s model run (swap scope 78%):** naive skyfield swap ≈27–32 s (2.5–3.0×, measured 31.9 s); swap+batching ≈19 s (4.3×); astropy-only batching ≈23 s (3.6×); + vectorize `celestial_positions` + cache catalog reads ≈12 s (6.5×); all transforms free = floor ≈10 s (8.1× max). Plain path: max 1.09×.

**Recommendation:** don't ditch astropy. (1) batch `azel2radec` into one call per epoch in `predict_model`/SOURCE table — or better, pass catalog `rarad`/`decrad` straight through and skip the az/el round trip entirely (`azel2radec` then only runs for Sun/Moon: 5760→120 calls); (2) skyfield is only worth adding for Sun/Moon (146×) and scalar calls (7×), and only in batched form; (3) cheap wins regardless of the decision: vectorize the client's per-satellite ITRS→ICRS loop (18 s/run) and cache the 3CRR/SUMMS/MKGains file reads (2.5 s/run).

**Bug found while profiling:** GNSS sources fetched via tart-catalogue-client **never reach the model or the MS SOURCE table** — `predict_model` filters `s.get('el', -90) >= 45`, but client sources carry `dec_degrees`, not `el` (and the client-side filter `np.degrees(s["dec"]) >= filter_elevation` treats declination as elevation). Measured 0 SOURCE rows / 0 GNSS sources predicted on 2026-06-12. The az/el embedded in `data_test.json` hides this in the tests. Separate fix needed — I'll file it unless someone beats me to it.

tmolteno, 2026-09-28:
Follow-up on the bug found while profiling: fixed in `7819f07` (pushed).

- `__fetch_sources_via_client` now fetches true topocentric Az/El via the catalogue client's `horizontal_positions()` and filters on real elevation (it was applying a declination cut masquerading as an elevation cut, and sources had no `el` key at all);
- the SOURCE-table writer uses `ra`/`dec` directly when present — same convention as `predict_model`, so SOURCE DIRECTION matches MODEL_DATA — and skips the azel2radec round trip;
- `test_elevation_filtering` (which had enshrined the declination semantics) now asserts true elevation filtering.

Verified end-to-end on the 2026-06-12 HDF with `--add-model`: **0 → 108 GNSS SOURCE rows** (18 sats above 45° × 6 catalog epochs) and a predicted GNSS model component where there was none; full suite 50/50. Model wall time 81.5 → 85.6 s (~5%, and that now includes 60 extra GNSS predict epochs + 108 SOURCE rows), `azel2radec` call count unchanged.

The profiling numbers above are unaffected — they were measured on the code where GNSS contributed nothing, so the batching recommendation stands as written.

tmolteno, 2026-09-29:
Implemented the profiling recommendations. Interleaved 3-round A/B on the shipped 60-ts test HDF5 (`--add-model`), baseline `7819f07` vs all changes:

| | before | after | |
|---|---|---|---|
| model wall (median) | 130.3 s | **20.7 s** | **6.3×** (6.2–6.7×/round) |
| `azel2radec` calls | 5760 | **120** | (60 celestial + 60 Sun/Moon epochs) |
| client `celestial_positions` (140 sats) | ~3 s | **45 ms** | |
| equivalence (MODEL_DATA + SOURCE, both test paths) | — | **bit-identical** (max abs diff 0.0) | |
| tests | — | tart2ms 50/50, client 17/17 | |

**Changes:**
1. **Batched `azel2radec`** (`a373bef`) — `predict_model` and the SOURCE-table writer now make one vectorized conversion per (source epoch, distance mode) instead of one scalar SkyCoord transform per source. Unit-level bitwise identical (point and distance modes).
2. **Catalog parsing cached + vectorized** (`7381372`) — parsed 3CRR/SUMMS/MKGains collections cached per (path, mtime, flux cut): repeat loads 3.58 s → 0.1 ms; the RA/Dec extraction is vectorized (`radec_rad`, ≤2.7e-15 rad fp reassociation vs the old per-source properties).
3. **Vectorized the client's `celestial_positions`** (tart-catalog-client-py, catalogue repo `bc94e9c`) — one ITRS(array)→ICRS transform for all satellites instead of one per satellite; 0 exact output mismatches over 140 live TLEs.

Measured 6.3× vs the ~6.5× Amdahl ceiling predicted above (batching alone was predicted ~3.6×) — on target. HOST caveat: absolute walls swing ~1.5× with CPU load (a calmer-load spot check of the *baseline* was 85.6 s, implying ~13 s unloaded for the final code, matching the prediction).

Explicitly **out of scope** here, per review: the catalog `rarad`/`decrad` pass-through that skips the az/el round trip entirely — it changes directions by ≤20.5″ (annual-aberration convention), so it deserves its own decision even though it is far below TART's resolution.

(The astropy-vs-skyfield verdict above stands: all of this was achieved with astropy alone. Skyfield remains only worth considering for Sun/Moon.)

tmolteno, 2026-11-01:
Closing: the actionable work here has shipped and the central question is answered with data.

**What landed (0.9.4):**
- `a373bef` — batched `azel2radec`: one vectorized SkyCoord transform per source epoch in `predict_model` and the SOURCE-table writer (was one scalar transform per source, ~70% of model runtime);
- `7381372` — 3CRR/SUMMS/MKGains parsing cached per (path, mtime, flux cut) and vectorized RA/Dec extraction;
- `7819f07` — GNSS sources reach the model and SOURCE table again (the declination-as-elevation filter bug found while profiling; its follow-up is #68);
- vectorized `celestial_positions` upstream (catalogue `bc94e9c`).

Net on the shipped 60-timestamp test HDF5 with `--add-model`: **130.3 s → 20.7 s (6.3×)**, MODEL_DATA + SOURCE bit-identical, against a ~6.5× Amdahl ceiling from the profiling — i.e. essentially all of the predicted win, with astropy alone.

**Verdict on the original ask:** don't ditch astropy. Like-for-like, skyfield wins on scalar azel→radec (7×) and Sun/Moon (146×), but loses on the vectorized radec→azel path (astropy 10–14× faster) — and after the batching work the scalar call shape no longer exists in tart2ms. Plain conversions were never the bottleneck (max 1.09×).

**Deliberately left out (tracked elsewhere, not needed now):**
- the catalog `rarad`/`decrad` pass-through that skips the az/el round trip: changes directions by ≤20.5″ (annual-aberration convention) — that's a decision under #56, not a performance one;
- a batched skyfield fast-path for Sun/Moon only: 146× on that one transform but only ~5% of model runtime — not worth a new dependency.

Closing as completed; if a future profile shows transforms dominating again, reopen or file fresh with numbers.

closed 2026-11-01 by tmolteno