tart-telescope/tart2ms #54: Capture 0-gain antennas and auto flag

bennahugo, 2023-10-18:
Currently zeroing the gains on the antennas do not raise flags. This causes gains subsequently derived with external tooling (e.g. CASA)  to blow up. We need to catch antennas that are turned off and flag them appropriately for the purpose of calibration and imaging.

tmolteno, 2023-12-10:
We could create an automatic flagger that uses the GPS correlation amplitude to flag antennas? This could be done as part of tart2ms, or alternatively we could just use the zero gain antennas and consider them flagged. 

* Immediate Solution: Auto flag zero gain antennas. (this means the cal decided they weren't operational
* Longer term solution: Generate an auto-flagger that can be used to flag antennas 
```tart_flag_data --target https://tart.elec.ac.nz/signal --use-gnss --ms mydata.ms

I like the latter solution as we can create a mini subproject that would be generally useful for student research... But the former will cause fewer rookie mistakes...

closed 2026-09-27 by tmolteno

tmolteno, 2026-09-28:
Implemented in commit `e4864c8` (pushed; auto-closed by the commit message). Antennas with zero (or non-finite) stored gains now get every baseline touching them written with FLAG set — detected in both the HDF5 and JSON loaders, including under `--uncalibrated` where the gain override would otherwise hide it, with a warning naming the antennas. Covered by a new end-to-end test (`test_zero_gain_antenna_flagged`) plus unit tests; full suite 50 passed.

Note the MS has no per-antenna flag column, so row flags are used — CASA derives per-antenna flags from those. One unrelated latent quirk noticed while in there: `cal_vis.set_gain(np.arange(24), gains)` hardcodes 24 antennas (tart2ms/tart2ms.py) — fine for TART, but worth a follow-up if non-24 arrays are ever converted.