tart-telescope/tart2ms #56: Barycentric -> Geocentric conversions when rephasing celestial coordinates

bennahugo, 2024-04-12:
Although still well within the instrumental resolution there is a problem in using current ICRS celestial coordinates and rephasing far off the original zenith position in celestial coordinates. If we build larger TARTs this will become more of an issue, so I'm documenting this now.

The UVW bases are in geocentric frame, which means the l,m and n coordinates should also be in geocentric frame. However, we are using the ICRS coordinates in computing the cosines, which are in barycentric reference frame.

See discussion in [codex-africanus#](https://github.com/ratt-ru/codex-africanus/issues/305).

I don't think it really affects things that are in orbit of course, including the model we use to ultimately calibrate phases, but if we are tracking a distant object one day this will be an issue in the astrometry

tmolteno, 2026-09-28:
Confirmed still present in master, and quantified — the description is accurate.

**Where the frames mix:** uvw is geocentric (casacore synthesizes it from ITRF baselines, tart2ms/fixvis.py) while the l,m,n used for rephasing are built from barycentric ICRS direction differences (astropy ICRS zenith/catalog positions in tart2ms/tart2ms.py and fixvis.py), and the MODEL_DATA path shares the assumption via codex-africanus `radec_to_lm` (ms_helper.py) — the ratt-ru/codex-africanus#305 case.

**Impact:** dominated by annual aberration (≲ 20.5″, largely a uniform ~1e-4 scale on the image plane) plus diurnal aberration (≲ 0.3″) → ≲ 7e-3 rad phase error on TART's 2 m max baseline at L-band and ~1e-4 position-scale error — far below TART's ~40′ resolution, as expected from "well within the instrumental resolution".

**Proper fix (when needed):** transform the reference/catalog directions to geocentric apparent (astropy GCRS at obstime — aberration, plus parallax for solar-system bodies) before computing direction cosines in `rephase()` and `radec_to_lm`, or equivalently synthesize uvw in the barycentric frame delay-model style — keeping rephase, the africanus predict and the casacore uvw in one convention. Suggest documenting the approximation and deferring until larger arrays/wide-field work demand it; happy to implement the GCRS version when that day comes.

tmolteno, 2026-11-06:
## Update (2026-11-06): the catalogue-client now supplies the geocentric directions the proper fix needs

**What changed upstream**

- `tart-catalogue-client` **0.6.5** (released today) fixes tart-telescope/catalogue#14: `celestial_positions()` no longer returns *barycentric* ICRS directions. It now returns **GCRS (geocentric)** directions by default, or **topocentric** RA/Dec when `lat`/`lon` are given (a pure rotation of the observer baseline — no barycentric offset reintroduced). tart2ms now floors at `>=0.6.5` with a regression test that pins non-collapsed satellite directions.
- tart2ms's UVW synthesis (`fixvis.synthesize_uvw`) was reimplemented on erfa + astropy this week (python-casacore removed): the frame is the ITRS→ICRS chain (CIO precession-nutation, Earth rotation, polar motion) with the source placed at its *apparent* place (aberration, light deflection), validated against python-casacore 3.8.1's `to_uvw()` to **<~1.7 arcsec** of frame rotation (~1e-5 m on TART's baselines), with a golden test pinning the old output.

**Where the frames still mix (unchanged from the analysis above)**

- `rephase()` (fixvis.py) computes ll/mm/nn from J2000 RA/Dec deltas;
- `predict_model` (ms_helper.py) feeds J2000 `sources_radec` + `zenith_directions` to codex-africanus `radec_to_lm` (ratt-ru/codex-africanus#305);
- the SOURCE table keeps J2000 via the azel2radec path (#68 stays as is — that frame is what the table wants).

**Path to the fix, now that the dependency exists**

Transform the phase centre and the catalog directions to GCRS at obstime before the direction cosines: the catalog side can be taken directly from `celestial_positions()` (GCRS, or topocentric with lat/lon); the phase centre needs an astropy GCRS transform of the FIELD PHASE_DIR per epoch. After this week's UVW change the w-axis is already the geocentric-apparent direction, so the residual is precisely *barycentric l,m,n vs geocentric-apparent w* — one transform on each side of `rephase()`/`radec_to_lm` closes it. Impact stays as quantified above: ≲20.5″ (annual aberration) → ≲7e-3 rad phase error on a 2 m baseline, deferred until wider-field work demands it.